victronenergy/gui-v2 #1439: Double check that inverter warnings cause notification be triggered

chriadam, 2024-09-05:
This discussion was triggered by https://victron.discourse.group/t/3-50-13-gui-v2-low-battery-warnings-not-signaled/521
Which is slightly different, as it's about LED indications of low battery.
But Matthijs wants to double check that inverter warnings to indeed trigger notification page.

vejpasop, 2024-09-05:
The difference between GUI V1 and V2 is mostly the low battery led indication in the Multi

On GUI V1 in case of a low battery **warning** you will see the low batter led flashing on GUI. 
[image]

Since it is an warning the GUI will not jump to notifications. It will not jump to the notification screen. Both GUI V1 and V2 will only do that in case of an **alarm**

vejpasop, 2024-09-05:
On GUI V2, when there is an alarm the user interface jumps to the Notification screen and the backlight comes on. Also when the alarm is cleared the backlight comes on. On a warning the backlight does not come on and the user interface does not jump to the notification screen. You will not be informed about the warning unless you touch the screen.
Example: 
Screen1,low battery warning active 
[image]

Screen 2. when you touch the screen the red dot on the Notifications page selector only hints about the warning
[image]

vejpasop, 2024-09-05:
Note. When Matthijs was testing on my Ekrano he needed to touch the "Acknowledge alerts" button in the notification screen twice in order to make it respond. He could reproduce it. However when I try I can't reproduce it on the same Ekrano.

chriadam, 2024-09-11:
Thank you for testing!  Created #1455 to track that latter issue.  Closing this issue, as it seems that according to the testing it works as required.  Please reopen if I am mistaken.

closed 2024-09-11 by chriadam