victronenergy/gui-v2 #1442: GUI V2 3.50~15 Lynx Shunt peukert exponent still cannot be changed

DanielMcInnes, 2024-09-05:
see https://victron.discourse.group/t/gui-v2-3-50-15-lynx-shunt-peukert-exponent-still-cannot-be-changed/1251/3

martinbosma, 2024-09-09:
@DanielMcInnes I will arrange a test setup

martinbosma, 2024-09-10:
@DanielMcInnes See https://vrm.victronenergy.com/installation/8072/dashboard. I've sent an invitation just now.
Feel free tho change any setting on the Lynx Shunt, it is not connected to anything.
[image]

I've also noticed that the device list entry shows -- for the Soc in the summary, while on gui-v1, this value is hidden. If this is by design, then it's fine by me. But otherwise perhaps copy the behavior from gui-v1.
[image]
[image]

commit a95cf0fc (2024-09-24): Change types of 'from' and 'to' to 'real' in NumberSelectorDialog and ListSpinBox

Fixes #1442
commit e916e35e (2024-09-30): Change types of 'from' and 'to' to 'real' in NumberSelectorDialog and ListSpinBox

Fixes #1442
PR #1499 (2024-09-30): Change types of 'from' and 'to' to 'real' in NumberSelectorDialog and…
… ListSpinBox

Fixes #1442
commit 1263d4e9 (2024-09-30): Change types of 'from' and 'to' to 'real' in NumberSelectorDialog and ListSpinBox

Fixes #1442