victronenergy/gui-v2 #1446: GUI v2 missing from Display Settings

seamaster101, 2024-09-07:
I installed the newest beta 3.5~20
gui v2 is available remotely "venus.local/gui-beta" no problem, but I can not enable the GUI V2 to display on the screen - VenusOS large V3.5~20 beta running on RPI3B. 
There is no menu available to switch to gui v2 in settings>display&language
What is going on? How to display the new guy on the screen?

carlosgaros, 2024-10-15:
Also please give me the option to enable it. Many of us have screens. I think we can contribute a lot by testing. Or please give me instructions on how to enable it from the terminal.

mpvader, 2024-10-28:
Hi, its impossible at the moment; the problem is missing or not working graphics acceleration driver in venus os for the pis. 

So, needs work by one of the gurus

Webifi, 2025-02-15:
@mpvader  Some clarity on the status here would be helpful.

In [this announcement](https://community.victronenergy.com/t/venus-os-v3-50-official-release-new-user-interface-security-and-generator-improvements/9588), it's stated that gui v2 not working on pi is a known issue, but also "good news is that it is being worked on", linking to a thread that is hidden or missing, so no real details.

In [this thread](https://community.victronenergy.com/t/raspberrypi-gui-on-screen-is-still-old/18055/3), it seems you're basically say "not my problem", pushing it off for others to investigate.

So, is it being worked on, or is it dead in the water?