victronenergy/gui-v2 #1447: GUI v2 stuck in rebooting screen

wetty, 2024-09-08:
I am running venus Os 3.50-20 on a Raspberry 3+

Since a few versions the GUI v2 is always showing the following screen. Rebooting or updating to a newer Venus OS does not change ist. Tested with several Browsers on several screens. It seems this happens only on my local network. Accessing the gui-v2 via https://192.168.x.x/gui-beta/
[image]

In the VRM Console the GUI v2 is correct

[image]

chriadam, 2024-09-11:
Thanks, I will fix this.

chriadam, 2024-09-11:
Oh, this is a duplicate of issue #1445 - closing.

chriadam, 2024-09-11:
Ah, no, slightly different (but related) case.  Reopening.

chriadam, 2024-09-16:
I was unable to reproduce this issue.  I have merged into main a fix which I think resolves the issue, but without being able to reproduce I am not entirely certain.  Closing this issue for now - please reopen if you can still reproduce with gui-v2 v0.3.11 or later.

closed 2024-09-16 by chriadam