victronenergy/gui-v2 #1480: Reconnect does not work

mr-manuel, 2024-09-17:
When you have opened the WASM build and reboot the GX device, the GUI does not reconnect and the page need to be refreshed.

In the developer console of the browser you see that two retries are made and after a error is thrown.

```
BackendConnection state: Victron::VenusOS::BackendConnection::Reconnecting (2) venus-gui-v2.js:8:18090
Firefox cannot establish a connecton to the server at ws://venus.local/websocket-mqtt. venus-gui-v2.js:8:243746
```

[image]

chriadam, 2024-09-24:
Fixed.  Please let me know if you still see this issue when using v0.3.12.

closed 2024-09-24 by chriadam

mr-manuel, 2024-09-24:
Can confirm that this is fixed.