victronenergy/gui-v2 #1485: allow installers to adjust temperature calibration

DanielMcInnes, 2024-09-18:
as per gui-v1 commit 270b8704b75f3ae23ff7014424857c278887cfdd
the 'writeAccessLevel' needs changing

    allow installers to adjust temperature calibration
    
    If a temperature measurement deviates, installers now have the ability
    to compensate it. E.g. to correct the calibration of the on-board Cerbo
    temperature sensors. This used to be available to super users only.
    
    https://github.com/victronenergy/venus/issues/1182

mpvader, 2024-09-19:
super simple? then pls v1.0.0 and medium prio

DanielMcInnes, 2024-09-25:
No changes are needed for gui-v2 for this ticket, the default permission for ListSpinBox is already set to 'installer'

closed 2024-09-25 by DanielMcInnes