victronenergy/gui-v2 #1491: Always show battery SOC percentage in center of brief page

mpvader, 2024-09-19:
As its now, its only shown if the brief page only has one item. And hidden when there are two or more circle gauges. Further input here:

https://community.victronenergy.com/t/gui-v2-on-v3-50-7-can-we-have-soc-in-the-circle-on-the-brief-page/124

Next step: @jepefe will discuss this with Serj

chriadam, 2024-12-13:
I have created  a prototype (see the above PR).  It supports showing a center-value in 1 or 2 or 3 gauge case, but NOT in 4 gauge case (there's not enough space in the middle).  The 1-gauge case is unchanged from the current implementation (except that you can now choose to show a temperature value there instead of the battery state of charge value, as is currently supported).  There are settings to control: 1) which (battery or temperature sensor) device whose value should be shown in the center of the brief page, 2) whether to show the label above the value or not, 3) when the center value should be shown (e.g. only 1 gauge case, vs 1 or 2, vs 1 or 2 or 3 gauge case).  Please see the attached screenshots for the various 2 and 3 gauge cases, both with and without labels.  @Serj do you have any thoughts on this prototype?  Are there any specific changes you would make, or ways this could be improved?  Matthijs asked me to speak to you to get some proper design input so that we can iterate the prototype into a proper solution.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

chriadam, 2024-12-13:
Screenshots above are the 5 inch screen resolution; the 7 inch screen resolution is more spacious.
I tested with 100% and 100 C values, also, and they do fit (although only just) in the space, without any overlaps, in the 3 gauge case.

mpvader, 2025-01-20:
@chriadam was being able to show two battery banks on the levels also part of this issue / prototype? I remember discussing that, but am not sure where that ended up.

Also, while searching I saw a few more issues, ie #541 , and now #1857.

For me, it would be nice to do all of them in one go while at it.

chriadam, 2025-01-21:
I'm not sure about the "two different batteries" gauges thing - I'd have to talk to Bea to find out whether the information we use (system battery info) aggregates data already, or whether it's just a single battery.

I agree that 541 and 1857 can be implemented as part of the task, yes.

mpvader, 2025-01-30:
@chriadam / @blammit how about the two different battery gauges thing?

Aggregrating battery data is not how I’d like it to work.

blammit, 2025-01-30:
There is the system battery information under system/Dc/, and then there are the `battery` services.

Normally we just show the system battery information. If we want to show multiple batteries in the multi-gauge case, then what SOC and voltage/current/power would be shown in the centre? Would those still be the numbers from the system battery data, rather than individual battery services?

blammit, 2025-02-17:
Taking this as discussed with Chris.

@mpvader if we support displaying multiple batteries, that needs a bit more discussion. For example:

- If there are multiple batteries, does the system battery information feature anywhere on the Brief page at all? (The system battery data is what we show in the Overview page at the moment, so perhaps it would be odd if the Brief and Overview pages did not show the same battery data.)
- If there are two batteries, would one be for the centre information and the other for the outer gauge? Or maybe they are always circular gauges, and then the system battery data can always be shown in the centre for consistency.

And does this only apply to batteries? For example, do we need to support showing individual fuel tanks as well, instead of only showing one gauge with the aggregated fuel tank data in the case there are multiple fuel tanks?

blammit, 2025-02-28:
>  how about the two different battery gauges thing?

Discussed in the dev meeting yesterday that the feature to show individual batteries and tanks is separate, so I have created this task for it: https://github.com/victronenergy/gui-v2/issues/1949

closed 2025-03-06 by blammit