victronenergy/gui-v2 #1494: navigation in the brief page gets stuck due to a firmware update

jhofstee, 2024-09-19:
Seen on an Ekrano with gui-v2 version v0.3.11.

To reproduce run e.g. `/opt/victronenergy/swupdate-scripts/check-updates.sh -update -force`

jhofstee, 2024-09-20:
I had a look at this today to see if I could find what went wrong. Oddly enough I can't reproduce it, while yesterday it happened every time.

chriadam, 2024-09-24:
I have reproduced a similar issue, due to a crash in QMqttClient dereferencing a ptr to a deleted transport device (it doesn't properly guard the pointer passed via setTransport() even though it doesn't take ownership), resulting in the UI basically hanging.  The fix for #1480 fixes that problem, amongst a couple of other issues related to reconnection.

I suggest we close this issue (assuming that the fix for #1480 does in fact fix this one) and reopen it if someone is able to reproduce after the fix goes into main.

jhofstee, 2024-10-16:
yes, lets close it, and if it is reproducible it can be reopened.

closed 2024-10-16 by jhofstee