victronenergy/gui-v2 #1496: Allow start page to be configured

blammit, 2024-09-23:
Allow a "start page" to be configured. This is the page that will be shown:

- on application start-up
- after the application is idle for a certain time, as configured in settings

Add a settings page at Settings > Display & Language > Start Page, with these options:

- Start page
    - Auto  <-- this is the default selection
    After one minute of inactivity, if the current page is one of the following, then save it as the start page.
    - Brief
    - Brief - side panel open
    - Overview
    - Levels - Tanks
    - Levels - Environment
    - Battery overview
    
- Timeout
  Revert to the start page when the application has been idle for this duration.
    - Never  <-- this is the default selection
    - 1 minute
    - 5 minutes
    - 10 minutes
    - 30 minutes
    - 60 minutes

blammit, 2024-09-23:
Added "blocked" label, pending some settings to store these values, e.g. `/Settings/Gui/StartPage` and `/Settings/Gui/StartPageTimeout`.

jhofstee, 2024-09-24:
I added them as /Settings/Gui2/StartPage /Settings/Gui2/StartPageTimeout so that it is clear they belong to gui-v2 only.

blammit, 2024-09-24:
Thanks! Removing blocker tag.

blammit, 2024-09-24:
Thanks, removing blocked status.

blammit, 2024-10-17:
> Start page
> Show this page on application start-up.
>
> Auto <-- this is the default selection
> After one minute of inactivity, if the current page is one of the following, then save it as the start page.

Actually, the "auto" option should be a separate toggle, not an extra option in the list. Otherwise, using the "auto" feature means that a page would become the start page automatically, but then it will not change again after that (unless the user goes to the settings to change it), which would be confusing. Will update the task.

commit 2e95cd31 (2024-10-16): Add configurable "start page"

Allow a "start page" to be configured from Settings > Display &
Language > Start Page. This is the page that will be shown:

- on application start-up
- after the application is idle for a certain time

If the "Auto" option is selected, and a timeout is set, then the start
page will be set automatically from the current application page after
one minute, if it is one of the available options.

Fixes #1496
commit f428718d (2024-10-16): Add configurable "start page"

Allow a "start page" to be configured from Settings > Display &
Language > Start Page. This is the page that will be shown:

- on application start-up
- after the application is idle for a certain time

If the "Auto" option is selected, and a timeout is set, then the start
page will be set automatically from the current application page after
one minute, if it is one of the available options.

Fixes #1496
commit 3df7d365 (2024-10-21): Add configurable "start page"

Allow a "start page" to be configured from Settings > Display &
Language > Start Page. This is the page that will be shown:

- on application start-up
- after the application is idle for a certain time

If the "Auto" option is selected, the start page is set using the
current application page after one minute of inactivity, if the
current page is one of the available options.

Fixes #1496
commit 610889e0 (2024-10-21): Add configurable "start page"

Allow a "start page" to be configured from Settings > Display &
Language > Start Page. This is the page that will be shown:

- on application start-up
- after the application is idle for a certain time

If the "Auto" option is selected, the start page is set using the
current application page after one minute of inactivity, if the
current page is one of the available options.

Fixes #1496
closed 2024-10-23 by blammit