victronenergy/gui-v2 #1503: Fix 'ListLabel' binding loop

DanielMcInnes, 2024-09-25:
qrc:/qt/qml/Victron/VenusOS/components/listitems/ListLabel.qml:13:2: QML ListLabel: Binding loop detected for property "height"

MikeTrahearn-Qinetic, 2025-02-27:
Do you have a specific reproduction case? Because there are a number of such cases around the system - for example, one I noticed was in mock mode, open the notifications page. There are three mocked up notifications there. Click on one to have it move to the history view and you get a binding loop on "height" occurring there. This could be quite complex to unravel but needs to also consider carefully case by case as the introduction of VisibleItemModel has added more potential layers of bindings for "visible" in some cases, but in others for example, the delegate is now no longer required so it disappears. So some of the binding loops might have gone away and we might now have new ones.

chriadam, 2026-03-02:
I believe this will be automatically fixed by https://github.com/victronenergy/gui-v2/pull/2808 but we need to double check once that one is merged.