victronenergy/gui-v2 #1509: Translations broken

DanielMcInnes, 2024-09-26:
see https://community.victronenergy.com/t/v3-50-24-gui-v2-some-chinese-strings-not-translated/4917

jhofstee, 2024-09-26:
related https://github.com/victronenergy/gui-v2/issues/1256

dirkjanfaber, 2024-09-26:
In the translation file `i18n/venus-gui-v2_zh_CN.ts` , I do see the translation being available:
```
    <message id="common_words_battery_temperature">
      <location filename="../../components/CommonWords.qml" line="86"/>
      <source>Battery temperature</source>
      <translation>电池温度</translation>
    </message>
```

DanielMcInnes, 2024-09-26:
> In the translation file `i18n/venus-gui-v2_zh_CN.ts` , I do see the translation being available:
>
> ```
>     <message id="common_words_battery_temperature">
>       <location [...]

Yes, I think something else is going wrong here

DanielMcInnes, 2024-09-26:
I remember seeing something in the console logs along the lines of: "Error: ZH_cn country code not recognized", which could have something to do with it

blammit, 2024-09-27:
This is the same issue as https://github.com/victronenergy/gui-v2/issues/1234?

DanielMcInnes, 2024-09-27:
It looks like the same issue

DanielMcInnes, 2024-09-30:
dmcinnes@dmcinnes-System-Product-Name:~/git/victron/desktop$ make venus-gui-v2_lupdate
...
File /home/dmcinnes/git/victron/desktop/i18n/venus-gui-v2_zh_CN.ts won't be updated: it contains translation but the target language is not recognized
...
Built target venus-gui-v2_lupdate
dmcinnes@dmcinnes-System-Product-Name:~/git/victron/desktop$

dirkjanfaber, 2024-09-30:
The full discussion is on [slack](https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1708496412587859), but in short, the part behind the underscore gives more info on the language, as there are more dialects within the language.

Simplified=zh_CN and Traditional=zh_TW. 

As we are using simplified Chinese, we use the `_CN`. 
 
 Also see https://github.com/victronenergy/gui-v2/pull/788.

closed 2024-10-02 by DanielMcInnes