victronenergy/gui-v2 #1510: Hide invalid motordrive items

philipptrenz, 2024-09-26:
Depending on wether the "Motordrive" service is provided by a 3rd party motor integration or a SmartShunt, some parts may not be available. I updated the [dbus motordrive service specification](https://github.com/victronenergy/venus/wiki/dbus#motordrive), please hide all optional paths in case they are invalid.

[image]

blammit, 2024-10-02:
@philipptrenz can all paths be hidden if they are invalid? If not, which ones are considered optional?

philipptrenz, 2024-10-02:
@blammit We're currently working on a new motordrive integration and it questions the optional paths currently marked as optional in the dbus specification. 

Therefore: Let's do it as you suggested, hide any path if invalid 👍

blammit, 2024-10-02:
Will do, thanks 👍

commit b3346e94 (2024-10-08): Hide invalid Motordrive items

Fixes #1510
closed 2024-10-08 by blammit