victronenergy/gui-v2 #1522: Lynx parallel BMS on WASM not displaying system voltage when BMS is pending

ReinvdZee, 2024-09-30:
If one of the paralleled BMSes is 'pending' (waiting until its voltage level is in range with the system voltage), we show the system voltage in the menu of that BMS. However, on WASM, this field is not set properly:

[image]

On the local UI, the field is showing the value correctly.
See [original issue](https://github.com/victronenergy/gui-v2/issues/1269)

DanielMcInnes, 2024-10-07:
Hi @ReinvdZee , I will need some help to debug this one. Are you available some time this week?

DanielMcInnes, 2024-10-16:
dbus path:
dbus/com.victronenergy.battery.lynxparallel1/Dc/0/Voltage

mqtt path: 
mqtt/battery/1/Dc/0/Voltage

commit 57f52eba (2024-10-16): Fix path for Lynx parallel BMS system voltage

Fixes #1522
commit dd8ac8ef (2024-10-16): Fix path for Lynx parallel BMS system voltage

Fixes #1522
commit 6dc2c2f0 (2024-10-16): Fix path for Lynx parallel BMS system voltage

Fixes #1522
commit 4c99e485 (2024-10-16): Fix path for Lynx parallel BMS system voltage

Fixes #1522
closed 2024-10-18 by DanielMcInnes