victronenergy/gui-v2 #1543: Not possible to configure high battery temperature alarm

RichardTsc, 2024-10-06:
In the gui-v2, I went through all menu items looking for a way to configure a high battery temperature alarm. I'm able to configure alarms for high/low tank levels and high/low temperature from my sensors, but not for the temperature sensor from the SmartShunt. 

Is this a bug or am I missing the point where I can configure this? To my understanding, the alarms configured in VRM are a different kind of alarms which also do not show on the gui-v2

DanielMcInnes, 2024-10-07:
Hi @RichardTsc , you should go back one level from there, and click 'Settings'
[image]

Then click 'Alarms' from inside the 'Settings' page
[image]

RichardTsc, 2024-10-07:
Hi @DanielMcInnes , thanks for the swift reply. Unfortunately, I'm missing exactly this "Settings" item between History and Device. Any idea why?

[image]
 
[image]

DanielMcInnes, 2024-10-07:
Hi @RichardTsc , I don't know sorry. The 'Settings' item is only shown if your smart shunt supports it, eg. from the path `.../battery/2/Settings/HasSettings`. I don't know why your SmartShunt doesn't support it. Try reading this: https://www.victronenergy.com/media/pg/SmartShunt/en/all-features-and-settings.html , it might be something you have to set up via VictronConnect

RichardTsc, 2024-10-16:
@DanielMcInnes, you are right, I was able to set it up via VictronConnect and it works on the gui-v2 after its set up there. 

Would argue, that it should be able to configure also without VictronConnect, as this is one of the bread and butter shunts and nothing special

chriadam, 2024-10-16:
Not a bug in gui-v2.  Perhaps a feature request for venus-platform?  Please make a post on the community forum: https://community.victronenergy.com/c/betas/venus-betas/18

closed 2024-10-16 by chriadam