victronenergy/gui-v2 #1549: add text to ess menu about rs settings location

DanielMcInnes, 2024-10-09:
as per gui-v1 commit 25ce2c3541358d84f07722a5309975bc22a85a22

    add text to ess menu about rs settings location
    
    Add text to the ESS menu that shows up if there is an RS system,
    telling the user that the ESS settings can be accessed from the
    device list.
    
    https://github.com/victronenergy/venus-private/issues/404

mpvader, 2024-10-21:
on my system, which has Multi RS, this either doesn't work, or I don't understand how its supposed to work, @izak 

[image]

https://vrm.victronenergy.com/installation/358999/overview

izak, 2024-10-21:
Aaah I know! Old wasm on VRM:

[image]

It works correctly with v0.3.15.

mpvader, 2024-10-22:
👍

closed 2024-10-22 by mpvader