victronenergy/gui-v2 #1552: Implement connectivity icons

ReinvdZee, 2024-10-09:
Implement the Wifi and cellular connectivity icons in the status bar, as per the design:

[image]

Not a blocker for V1.0.0.

DanielMcInnes, 2024-10-22:
Hi @ReinvdZee , I can't find these icons in the design, do you know where they are?

nmbath, 2024-10-22:
Would it be possible to also have an icon for ethernet being connected?

DanielMcInnes, 2024-10-22:
> @DanielMcInnes Icons are here: https://www.figma.com/design/GjCtz3A3jqSGvyjVvOm6RZ/GX-Touch-design?node-id=2110-140997&node-type=canvas&m=dev
> 
> But I think you added some components for the GSM settings a while back, to generate the icons for cellular strength? Don't remember if we added something similar for Wi-Fi settings as well.

thanks Bea

DanielMcInnes, 2024-10-22:
> Would it be possible to also have an icon for ethernet being connected?

We have a meeting later today, I'll bring it up then. @mpvader FYI

mpvader, 2024-10-31:
hey @nmbath 

> Would it be possible to also have an icon for ethernet being connected?

no; too much clutter.

blammit, 2024-11-12:
Setting medium priority for 1.1.0

commit 1ae3569d (2024-11-05): Add connectivity icons for wifi and gsm

Fixes #1552
commit c05e4803 (2024-11-05): Add connectivity icons for wifi and gsm

Fixes #1552
commit c6dd1475 (2024-11-05): Add connectivity icons for wifi and gsm

Fixes #1552
commit c0fcc79b (2024-11-25): Add connectivity icons for wifi and gsm

Fixes #1552
closed 2024-11-25 by DanielMcInnes