victronenergy/gui-v2 #1554: Genset total power not displayed on devicelist summary when `/Ac/Power` is not present.

ReinvdZee, 2024-10-09:
The summary of a connected genset device in the devicelist displays the state and the total power drawn. The model uses the path `/Ac/Power` to fetch the total power. However, some gensets do not support this field, and an invalid power value is displayed:

[image]

gui-v1 just sums the power values of the individual phases when this field is unavailable. V2 should do the same.

mpvader, 2024-10-17:
Hey @ReinvdZee when or by what services is this total power missing?

reason to ask it that I prefer to solve this there, rather than have to burden user interface with this.

Either we mandate having /Ac/Power for all things todo with AC; or the opposite: we remove it everywhere.

With everywhere I mean also beyond of the scope of gensets, ie this is also for Multis, chargers, and so forth. At the moment I don’t remember if we are going to the direction of adding it everywhere or the opposite; I’ll ask @izak .

ReinvdZee, 2024-10-17:
It was decided ([discussion on slack](https://victrondevelopment.slack.com/archives/C7C1ZU6UD/p1729141554596209)) that the drivers need to make sure that the /Ac/Power path is always there. So we won't solve this here.

closed 2024-10-17 by ReinvdZee