victronenergy/gui-v2 #1556: Genset UI: run time and service changes

DanielMcInnes, 2024-10-10:
as per gui-v1 commit 2437020904e7bc6e11c09bd49aab2a293ab73f01

this may already be complete

    Genset UI: run time and service changes
    
    - Remove Operating time from the Engine page.
    - Sync total run time with genset `/Engine/OperatingHours` (dbus_generator)
    - Add 'Run time and service' menu at top level
    - Remove Total run time from top level since its part of the run time and service menu.
    - Change 'Run time' to 'Current run time' on top-level
    - Only show 'Current run time' when generator is running, warming up or cooling down.
    - Move service interval setting to run time and service menu
    - Remove seconds notation from manual start toast since seconds cannot be specified.
    - Reordered menu items of generator start stop (relay).
    - Rename settings menu for DC gensets to "DC genset settings" because
    the auto start/stop settings was moved to the top level.
    
    https://github.com/victronenergy/venus/issues/1315

mpvader, 2024-10-10:
Rein will check if all this is already included in gui-v2, (as he expects), and if yes close it.

ReinvdZee, 2024-10-10:
This is included with [this PR](https://github.com/victronenergy/gui-v2/pull/1448), closing issue.

closed 2024-10-10 by ReinvdZee