victronenergy/gui-v2 #1561: Control panel: swiping to access cards on the right doens't work

mpvader, 2024-10-10:
Seen by Bob, in MS Edge, Google Chrome, en Firefox.

For myself it does work ok on the same site

Bob has seen this on this system: https://vrm.victronenergy.com/installation/3638/advanced

https://github.com/user-attachments/assets/8c4e06b8-848c-4e69-9c2d-6dc2a762aef8

mpvader, 2024-10-10:
an update on this: Bob reported that after refreshing his (MS Edge) browser it now works.

And also on all other browsers it works (he closed and reopened those browsers, his normal go to browser is MS Edge)

So, @chriadam , as far as I'm concerned we can close this issue. And for the future, I'll tell people to refresh their browser first; and if that solves issues, either this one or others, often then we need to look into something again.

chriadam, 2024-10-14:
Interesting.  Thanks!

closed 2024-10-14 by chriadam