victronenergy/gui-v2 #1563: Popup for "Detect generator at AC input" setting

ReinvdZee, 2024-10-10:
In gui-v1, this popup shows up, notifying the user why the setting can't be enabled:

ReinvdZee, 2024-11-25:
@blammit it would be great if you or someone else from Qinetic could take this one.

MikeTrahearn-Qinetic, 2024-12-12:
For my benefit: Go to Settings > Generator Start/Stop > Settings to get the equivalent GUI V2 page.

MikeTrahearn-Qinetic, 2024-12-12:
Reference code from gui (v1): https://github.com/victronenergy/gui/blob/master/qml/PageSettingsGenerator.qml#L51

MikeTrahearn-Qinetic, 2024-12-12:
Note: the MbSwitch overrides (shadows) its own enabled property here which is why on first glance it is difficult to understand how each toast is triggered when the MbSwitch is apparently disabled. In implementing this feature in gui-v2 we do NOT want to do any such property shadowing. For gui-v2 ListItems we have access to the content Row item which can be separately enabled from the rest of its parent item, so this should be the starting point for this fix.
However we should take into account such additional logic such as ListRadioButtonGroup which has enabled set on its root according to user write access. However we note that binding to the exposed inner content should still respect the enabled property of its parent, unlike MbSwitch here, which wouldn't and fundamentally breaks QQuickItem enabled principles.

MikeTrahearn-Qinetic, 2024-12-23:
On inspection the code already exists in Gui-v2 and has been copied almost verbatim from gui-v1 with only those changes required to suit gui-V2's API.
However - due to the MbSwitch shadowing the enabled property in gui-v1 and the ListSwitch in gui-v2 NOT shadowing the enabled property (because it is behaving itself), we have the silent issue now "correcting" the behaviour of the entire button to disable the entire button. So we now need to retro-fit this "wrong" behaviour.

MikeTrahearn-Qinetic, 2024-12-23:
We notice that gui-v1 nor gui-v2 would not consider the impact of clicking the button over and over if it never actually disables, creating multiple toasts...

MikeTrahearn-Qinetic, 2025-02-17:
This issue has been fixed as a result of #1898 which redefined the way ListItems work and now this button in question will not be disabled any more and this issue is no longer a problem.

commit b4f5d6c1 (2025-01-08): Show Popup for "Detect generator at AC input" setting

PageSettingsGenerator should correctly shows the existing toast by
changing the enabled property to content.enabled so that the button
may still be clickable. ListSwitch should be modified to emit its
clicked() signal when the content.enabled is false.

Fixes #1563
closed 2025-02-17 by MikeTrahearn-Qinetic