victronenergy/gui-v2 #1564: Manual control button switches to "Manual Stop", when generator is autostarted by a condition

ReinvdZee, 2024-10-10:
When the generator automatically starts by a condition, the user is not able to manually stop it. In gui-v1 the "Manual start" menu does not change when the generator starts by a condition. The user is able to "Manually start" the generator while it is already running by a condition. This overrides the reason it is running, so the generator will keep running, even if the condition that previously started it is resolved. 

V2 currently shows "Manual Stop" in the menu (also in the control card), which would normally only show if the user had manually started the generator.
[image]

So overriding the starting reason with a manual start is not possible in V2 currently.
Should we synchronize the behavior of V2 with V1?

jepefe, 2024-10-10:
The manual Stop button should only appear if the user has previously pressed the manual Start. It is acceptable, and actually a feature, that the user can press the manual Start even if the generator is already running due to another condition. In GUI-v1, we display a toast message informing the user that the generator will continue running if any active condition is present.

Active conditions take priority as the reason for starting or running the generator.

ReinvdZee, 2024-10-10:
Thanks @jepefe, we'll sync the behavior with V1 then.

kwindrem, 2024-10-10:
I've always found the interaction between manual and automatic confusing. One if the big issues is finding a way to stop the generator when it is started by an auto-run condition. The only way to do this is to turn off auto run which makes sense only if you understand the underlying mechanisms (code). Once in this condition, it's not possible to reenable auto run and have the generator remain stopped. If left in this condition, the generator will never start again automatically. Maybe what is needed here is "stop for xxx minutes" so that the automatic run conditions will be checked again in the future.

As described above, manual run overrides the automatic conditions so the generator will continue to run after all conditions have been satisfied. This is not explained well in the menus. Some improvement is needed here I think.

One thing I added to GuiMods at the request of a user was the ability to manually start the generator, then let the auto stop conditions stop the generator. For example if the SOC auto start is 20% and stop is 80%, manually starting with auto stop when the SOC is 50% will allow the generator to bring SOC to 80% then stop. (If SOC is at 90% already, manual start/auto stop won't do anything.)

closed 2024-10-21 by ReinvdZee