victronenergy/gui-v2 #1571: Control pane for ESS is missing for RS systems

mpvader, 2024-10-11:
See below for how it looks for VE.Bus systems. On an RS system, no entry in the control pane shows up. What is fundamentaly different here is that for VE.Bus systems, ESS control goes via `com.victronenergy.settings /hub4control`; and for RS systems, as well as any other new type Victron inverter/chargers, its in `com.victronenergy.acsystem/*`

First step here is @izak to outline exactly how this is supposed to work.

- [ ] Specify the details (dbus paths, etc). (@izak)
- [ ] Implementation (Qinetic)

[image]

mpvader, 2024-11-04:
There is a PR for this

blammit, 2024-11-07:
Fixed by https://github.com/victronenergy/gui-v2/pull/1681

closed 2024-11-07 by blammit