victronenergy/gui-v2 #1572: Starter battery of type com.victronenergy.genset displayed without any values

philipptrenz, 2024-10-11:
Most genset controllers provide a starter battery voltage reading. If it is available via `com.victronenergy.genset.*/StarterVoltage`, Venus OS seems to interpret the genset service as a battery. This is really helpful, as a low starter battery (voltage) causes a genset start failure.

However, gui-v2 does not yet seem to be able to interpret the service correctly. In the overview of all batteries in the system (via tap on the SystemBattery in the Overview), no helpful values are given for the genset starter battery.

[image]
[image]

For comparison: On VRM, the service is always called “Generator Starter Battery” and shows the only available value, the voltage. I would expect something similar on gui-v2.

[image]

chriadam, 2024-10-14:
@jepefe this might be something you need to discuss with Serj?  @blammit or is this feasible to do without any design?  Maybe for 1.1.0 not 1.0.0?

blammit, 2024-10-14:
@philipptrenz the battery list tries to fetch the battery data from the same paths, regardless of whether it is a `battey` or `genset`, etc. For example, looking at the `gdf` simulation in venus-docker, the `dcgenset` service there publishes values for `/Dc/0/Power`, `/Dc/0/Current` and `/Dc/0/Voltage`, so they appear in the battery list as expected. (However, there is no `/Soc` value, so the percentage on the right-hand side appears as "--".)

[image]

I'm not familiar with the DEIF genset controller but I assume it doesn't normally have those values? So maybe they need to show some different data, including `/StarterVoltage` etc.

@jepefe should `genset` and `dcgenset` services show a different set of data/metrics in the battery list, instead of showing voltage/current/power/temperature/soc like `battery` services?

blammit, 2024-10-14:
@chriadam Yes I think this makes more sense for 1.1.0, assuming this is not urgent. @philipptrenz

mpvader, 2024-10-14:
No design is needed here. Its a series of small changes in logic.

For batteries, of any of these parameters are not available, they need to be hidden rather than showing dashes:
- temperature
- current
- power
- state of charge

in case current isn’t available, the charging/discharging/idle indicator needs to be hidden as well

lastly, there is probably some logic to find the right path for the battery voltage for a genset service. I’ll find out and confirm here. Similarly, some batteries monitors can measure ome extra voltage.

Test site one:
[image]

from: https://vrm.victronenergy.com/installation/473156/overview

whats expected there is that all the dashes are hidden, also “idle” needs to be hidden, and for the top one show only a voltage.

Test site two:

[image]
[image]

From: https://vrm.victronenergy.com/installation/22539/overview

and here issue one is that that starter battery voltage is missing (there are three lines marked as visible in the config, yet only two show), and the other issue, aside from the dashes, is that the label isn’t taken. I labelled the Multi as “my multi”, yet the battery overview page those MultiPlus etc.

i’ll check where all that data is (I think in com.victronenergy.system /Batteries datastructure).

mpvader, 2024-10-14:
Here is the data. As I think we discussed before (there must be some misunderstanding here, as this is not the first time this comes up), this is the path to get the data from to populate the page showing all battery data.

And I'm 100% sure that this also solves the genset battery voltage missing, since VRM Portal works from exactly this data set.

```
root@nanopi:~# dbus -y com.victronenergy.system /Batteries GetValue
[{'active_battery_service': True,
  'current': 0.10000000149011612,
  'id': 'com.victronenergy.battery.socketcan_can0',
  'instance': 512,
  'name': 'Pylontech battery',
  'power': 5,
  'soc': 100.0,
  'state': 0,
  'temperature': 25.0,
  'voltage': 52.349998474121094},
 {'active_battery_service': False,
  'id': 'com.victronenergy.battery.ttyUSB1:1',
  'name': 'My starter votlage',
  'voltage': 0.029999999329447746},
 {'active_battery_service': False,
  'current': 0.0,
  'id': 'com.victronenergy.vebus.ttyS3',
  'instance': 275,
  'name': 'my multi',
  'power': 19,
  'soc': 100.0,
  'state': 0,
  'voltage': 52.31999969482422}]
root@nanopi:~# 
```

mpvader, 2024-10-14:
This is not a blocker for v1.0.0; nice to have.

and for v1.0.0 is a high prio

blammit, 2024-10-14:
> Here is the data. As I think we discussed before (there must be some misunderstanding here, as this is not the first time this comes up), this is the path to get the data from to populate the page showing all battery data.

> And I'm 100% sure that this also solves the genset battery voltage missing, since VRM Portal works from exactly this data set.

I see, thanks for clarifying. I thought the list of services was populated from `com.victronenergy.system/Batteries`, but that the data to be shown should still be fetched from the individual services. However, this is incorrect; both the services and the data should be generated using the value of `com.victronenergy.system /Batteries`, and there is no need to try to fetch data from the individual services.

philipptrenz, 2024-10-14:
@blammit 

> For example, looking at the gdf simulation in venus-docker, the dcgenset service there publishes values for /Dc/0/Power, /Dc/0/Current and /Dc/0/Voltage, so they appear in the battery list as expected.

This is not expected, as it's a generator and not a battery. But anyways, this will also be resolved as soon as `com.victronenergy.system/Batteries` will be used as a data source.

blammit, 2024-10-22:
@mpvader I noticed this one in the same `/Batteries` you provided:

```
 {'active_battery_service': False,
  'id': 'com.victronenergy.battery.ttyUSB1:1',
  'name': 'My starter votlage',
  'voltage': 0.029999999329447746},
```

This looks like a virtual battery, since there is no `instance` value? In this case I suppose it should not be possible to click on it and view more information (unlike other list entries you would want to click on the battery and see the detailed battery information, or the vebus details page if the battery comes from a vebus service, etc).

mpvader, 2024-10-28:
> This looks like a virtual battery, since there is no instance value? In this case I suppose it should not be possible to click on it and view more information (unlike other list entries you would [...]

correct @blammit  (sorry for the late reply, I had missed your previous one).

mpvader, 2024-10-28:
It looks like this is not solved correctly.

I've checked two systems, and its an issue only on the first one:
- https://vrm.victronenergy.com/installation/151755/dashboard
- https://vrm.victronenergy.com/installation/22539/overview (Pylon, here its OK)

json for the first one, not working:
```
root@einstein:~# dbus -y com.victronenergy.system /Batteries GetValue
[{'active_battery_service': True,
  'current': -9.39999961853027,
  'id': 'com.victronenergy.battery.socketcan_can1_vi0_uc167120',
  'instance': 0,
  'name': 'House battery',
  'power': 238,
  'soc': 95.0,
  'state': 1,
  'voltage': 26.4},
 {'active_battery_service': False,
  'current': 0.0,
  'id': 'com.victronenergy.battery.ttyUSB1',
  'instance': 289,
  'name': 'Service battery ',
  'power': 0.0,
  'soc': 100.0,
  'state': 0,
  'voltage': 13.479999542236328},
 {'active_battery_service': False,
  'id': 'com.victronenergy.battery.ttyUSB1:1',
  'name': 'Engine starter battery',
  'voltage': 13.6}]
```

json for the second one (Pylon), which works ok:
root@nanopi:~# dbus -y com.victronenergy.system /Batteries GetValue
[{'active_battery_service': True,
  'current': 0.0,
  'id': 'com.victronenergy.battery.socketcan_can0',
  'instance': 512,
  'name': 'Pylontech battery',
  'power': 0.0,
  'soc': 100.0,
  'state': 0,
  'temperature': 25.0,
  'voltage': 52.349998474121094},
 {'active_battery_service': False,
  'id': 'com.victronenergy.battery.ttyUSB1:1',
  'name': 'My starter votlage',
  'voltage': 0.029999999329447746},
 {'active_battery_service': False,
  'current': 0.0,
  'id': 'com.victronenergy.vebus.ttyS3',
  'instance': 275,
  'name': 'my multi',
  'power': 17,
  'soc': 100.0,
  'state': 0,
  'voltage': 52.279998779296875}]

screenshot of the 50ft yacht, where the  battery called "engine starter battery" is missing:

[image]

mpvader, 2024-10-28:
Do note: that 50ft yacht, the one not working, is a demo system. But, looking at the json I still don't see why gui-v2 shouldn't just display that starter battery

commit c10913d5 (2024-10-23): Battery list: show values directly from from system /Batteries

The /Batteries list provides values like "power", "voltage", and
"soc", so these can be used to show the values in the battery list.
This is better than assuming each /Batteries entry comes from a
service that provides paths like /Power, /Voltage, /Soc, etc. and
fetching the data values from those paths for each entry.

For example, the /Batteries list may contain a starter battery
provided by a genset service, and in this case the voltage is provided
by com.victronenergy.genset.*/StarterVoltage instead of /Voltage. The
backend us [... cut]
commit 16092d74 (2024-10-23): Battery list: show values directly from from system /Batteries

The /Batteries list provides values like "power", "voltage", and
"soc", so these can be used to show the values in the battery list.
This is better than assuming each /Batteries entry comes from a
service that provides paths like /Power, /Voltage, /Soc, etc. and
fetching the data values from those paths for each entry.

For example, the /Batteries list may contain a starter battery
provided by a genset service, and in this case the voltage is provided
by com.victronenergy.genset.*/StarterVoltage instead of /Voltage. The
backend us [... cut]
commit ce60f433 (2024-10-23): Battery list: show values directly from from system /Batteries

The /Batteries list provides values like "power", "voltage", and
"soc", so these can be used to show the values in the battery list.
This is better than assuming each /Batteries entry comes from a
service that provides paths like /Power, /Voltage, /Soc, etc. and
fetching the data values from those paths for each entry.

For example, the /Batteries list may contain a starter battery
provided by a genset service, and in this case the voltage is provided
by com.victronenergy.genset.*/StarterVoltage instead of /Voltage. The
backend us [... cut]
PR #1609 (2024-10-23): Battery list: show values directly from from system /Batteries
The /Batteries list provides values like "power", "voltage", and "soc", so these can be used to show the values in the battery list. This is better than assuming each /Batteries entry comes from a service that provides paths like /Power, /Voltage, /Soc, etc. and fetching the data values from those paths for each entry.

For example, the /Batteries list may contain a starter battery provided by a genset service, and in this case the voltage is provided by com.victronenergy.genset.*/StarterVoltage instead of /Voltage. The backend use [... cut]
closed 2024-10-29 by blammit