victronenergy/gui-v2 #1574: Solar from Inverter RS doesn't show in Brief/Overview and Device List

blammit, 2024-10-14:
When the solar comes from a RS Inverter, it does not appear in the Brief/Overview page; there is an empty space where the solar gauge and solar widget should be.

Also, in the Device List, the total power for the inverter is "0W", even though clicking on the device shows a non-zero value for the "PV" item.

Reproducible on https://vrm.victronenergy.com/installation/491097/overview

See slack thread: https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1728027496342769

blammit, 2024-10-15:
This is happening because the solar info is only shown on the Brief and Overview pages when there are `solarcharger` or `pvinverter` services present. On this installation, only `inverter` services are present.

mpvader, 2024-10-16:
Don’t forget .multi.

blammit, 2024-11-15:
Sorry, yes `multi` is already included as part of solar devices, forgot to put that in the comment.

blammit, 2025-01-09:
Updated the description with notes for investigation, tasks and testing/verification.

blammit, 2025-01-23:
Changes have been merged.

closed 2025-01-23 by blammit