victronenergy/gui-v2 #1578: Solar charger not showing active error

martinbosma, 2024-10-15:
See https://github.com/victronenergy/venus-private/issues/453

It appears the active error (`/ErrorCode`) is not shown on the Alarms & Errors page and the last errors (from history) are indicates as if there is an error. This is not true, although the active error is pushed into the last errors as soon as it occurs, but is never cleared.

Tested on a MPPT RS by inducing error 121.

On the device list summary it shows the error:
[image]

Nothing on the device page, but it falsely implies there are 4 active alarms/errors:
[image]

Alarms & Errors does not show active error, only the last errors from history:
[image]

martinbosma, 2024-10-15:
@blammit Can you fix the following items?
- Add the active error to the Alarms & Errors page?
- Drop the Last errors from Alarms & Errors
- Fix the "error counter" to only include active `Alarms/*` items and the `ErrorCode`
- Perhaps prefix the error code on the summary with a #, e.g. `Error: #121`

blammit, 2024-10-15:
@martinbosma should the "Last errors" really be dropped from Alarms & Errors? If we drop it, there is no other way to view the `/History/Overall/LastError*` for the charger.

I could just add the active error to the Alarms & Errors page, but without removing the Last Errors, but rename it to "Overall error history"? For example, here is an updated version, with:

- active error added to the Alarms & Errors page
- "error counter" changed to only include the active Alarms/* items and the ErrorCode

[image]
[image]

> Perhaps prefix the error code on the summary with a #, e.g. Error: #121

Sounds good, or alternatively we could show the full error description here instead, if that is preferable?

blammit, 2024-10-15:
> Perhaps prefix the error code on the summary with a #, e.g. Error: #121

@martinbosma  Sounds good, or alternatively we could show the full error description here instead, if that is preferable?

martinbosma, 2024-10-16:
>should the "Last errors" really be dropped from Alarms & Errors? If we drop it, there is no other way to view the /History/Overall/LastError* for the charger.

@blammit I discussed this with Matthijs yesterday and we decided to drop the last errors for now, until we have a better design.

>Sounds good, or alternatively we could show the full error description here instead, if that is preferable?

I'm afraid not all (translated) error descriptions will fit there. Do you have any insights in that? Otherwise just stick with the number.

blammit, 2024-10-17:
> @blammit I discussed this with Matthijs yesterday and we decided to drop the last errors for now, until we have a better design.

Okay, sounds good, thanks.

> I'm afraid not all (translated) error descriptions will fit there. Do you have any insights in that? Otherwise just stick with the number.

There is a bit of space available, but it doesn't wrap at the moment:

[image]

For 1.1.0 I'm planning some [reworking of the Device List implementation](https://github.com/victronenergy/gui-v2/issues/1339) to make custom per-service UI items, which will make it easy to wrap the error text, so we could add it at that point.

martinbosma, 2024-10-17:
>So I can just use the number for now. For 1.1.0 I'm planning some https://github.com/victronenergy/gui-v2/issues/1339 to make custom per-service UI items, which will make it easy to wrap the error text, so we could add it at that point.

Let look at this again when you are working on that issue.

commit 3c38cedd (2024-10-17): Show solar charger active error on drilldown page

Do not show the historical errors, as that is not relevant to this
page. Also when counting the number of errors and alarms, only
count the alarms and the active error, and do not count historical
errors.

Also add '#' to error code in the Device List.

Fixes #1578
commit 1e1a6f99 (2024-10-18): Show solar charger active error on drilldown page

Do not show the historical errors, as that is not relevant to this
page. Also when counting the number of errors and alarms, only
count the alarms and the active error, and do not count historical
errors.

Also add '#' to error code in the Device List.

Fixes #1578
closed 2024-10-18 by blammit