victronenergy/gui-v2 #1587: Increase sensor value limit precision

ReinvdZee, 2024-10-16:
Currently, sensor value limits can be specified with 1 digit of precision:
 
[image]

This should be increased to at least 2 digits or maybe even 3 digits. 
The smallest step the analog inputs can take is 0.005V. 

I think 2 digits for setting the sensor limits is sufficient. @mpvader?

mpvader, 2024-10-16:
Lets change it to two digits

mpvader, 2024-10-16:
From the user:

> Safiery Ultrasonics attached to a GX tank.
> 0 - 4v at maximum tank height (1 metre) but on a smaller tank it’s less. On my tank the range is 0.6v
> The GX Tank reads in .005 volt steps. It would be nice to be able to configure in those steps, or simply a button that says “set min” or “set max” which reads the current value to 3 decimal places and sets the min/max.
> I’ve had to use dbus-spy to get mine configured correctly.

so, on a second thought,  why not simply allow setting it in steps of 0.005V?

Its a bit extreme, but as long as the UI is smart enough wrt accelerating when keeping the button pressed, or add those set min and set max options, then I don’t see why not.

https://community.victronenergy.com/t/bug-sensor-value-a-writeable-field-in-v3-50-24/5179

commit 426bfc7c (2024-11-14): Increase sensor value limit precision

Change Sensor value limit precision (when full and empty) to 3 decimal places
and the stepSize to 0.005

Fixes #1587
commit f22040cf (2024-11-18): Increase sensor value limit precision

Change Sensor value limit precision (when full and empty) to 3 decimal places
and the stepSize to 0.005

Fixes #1587
commit d33757d8 (2024-11-20): Increase sensor value limit precision

Change Sensor value limit precision (when full and empty) to 3 decimal places
and the stepSize to 0.005

Fixes #1587
commit 1afcfddd (2024-11-25): Increase sensor value limit precision

Change Sensor value limit precision (when full and empty) to 3 decimal places
and the stepSize to 0.005

Fixes #1587
closed 2024-11-25 by MikeTrahearn-Qinetic