victronenergy/gui-v2 #1593: Add EVCS control card

blammit, 2024-10-17:
Add a control card for each EVCS service. It should show similar data to the control card in VRM:

[image]

MikeTrahearn-Qinetic, 2024-11-12:
Should there be one control card for each EVCS?

blammit, 2024-11-12:
> Should there be one control card for each EVCS?

Yes.

MikeTrahearn-Qinetic, 2024-11-14:
@blammit  There is quite a lot that can be done here and I have gotten a new card with a few test control working on it already. However we should decide on actual UI design for this based on existing UI precedents, for example the Inverter/Charger Control card and its various controls and the existing EVCS Settings controls.

MikeTrahearn-Qinetic, 2024-11-18:
Screen shots of work in progress - EVCS Cards are now working showing the name, status, mode, current and charging; current is only enabled when the mode is manual as per the EvChargerPage.

[image]
[image]
[image]
[image]
[image]
[image]

commit 3d089b34 (2024-11-25): Add EVCS Control Cards

EVCS Control Cards are added in the the Control Cards Page having feature
parity with the settings available from the Overview pages.

Fixes #1593
commit b0139ce3 (2024-11-25): Add EVCS Control Cards

EVCS Control Cards are added in the the Control Cards Page having feature
parity with the settings available from the Overview pages.

Fixes #1593
commit 4286124c (2024-11-25): Add EVCS Control Cards

EVCS Control Cards are added in the the Control Cards Page having feature
parity with the settings available from the Overview pages.

Fixes #1593
commit d0a3ce0f (2024-11-25): Add EVCS Control Cards

EVCS Control Cards are added in the the Control Cards Page having feature
parity with the settings available from the Overview pages.

Fixes #1593
closed 2024-11-25 by MikeTrahearn-Qinetic