victronenergy/gui-v2 #1595: Adjust conditions for showing arrow icon for positive/negative AC power on Brief/Overview

blammit, 2024-10-18:
When the active AC input's power is negative, show the green arrow (pointing left), and don't show a minus symbol.

When the power is positive, only show the blue arrow (pointing right) if ESS feedback is enabled.

Also, update the "ESS feedback to grid" check in SystemSettings.qml to simply check for  `com.victronenergy.system/Ac/ActiveIn/FeedbackEnabled = 1`. This allows the check to work on RS systems as well, unlike the current implementation that checks the value of `/Hub4`, `/Settings/CGwacs/OvervoltageFeedIn` and `/Settings/CGwacs/PreventFeedback`.

mpvader, 2024-10-18:
https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1729160215114419

blammit, 2024-10-29:
Changes merged.

closed 2024-10-29 by blammit