victronenergy/gui-v2 #1612: Incorrect firmware version of CAN connected energy meter

dirkjanfaber, 2024-10-23:
The running firmware on the energy meter is v1.08. Which is shown incorrectly when checking via _Services -> VE.Can port -> Devices -> Warmtepomp HQ2403FGD9C [43059]_  as version v1.06. 

[image]

Via the device list overview, the version is shown correctly as version v1.08. 
[image]

Also on the dbus, the version is show as v1.08.
[image]

Checked on system `416974`.

chriadam, 2024-11-01:
I don't have access to that system, but I found a system with a similar setup: https://vrm.victronenergy.com/installation/473156/
However, I couldn't reproduce the problem on the above system.

There is something worth noting: the FirmwareVersion value from the DeviceList appears to be a string (v1.08) according to the dbus-spy screenshot from Dirk-Jan above.  I wonder what the value is, under the vecan device path?

In the 473156 system the value is a number: `mqtt/vecan/vecan0/Devices/00002CC6E41E/FirmwareVersion = 1.08`

We do have some logic to convert numeric version to string (basically, just prefixing with "v" if the number is simple enough, but there are some rather complex rules for other cases, whose logic was copied directly from gui-v1 and I don't completely understand.  It's possible that for the specific device in question, one of those complex codepaths is being hit, but it seems unlikely (1.06 seems of the same family as 1.08 after all...)

Anyway, in short, I cannot reproduce, unfortunately.

dirkjanfaber, 2024-11-01:
At the moment, I am no longer able to reproduce the issue. The system in question did receive the 3.50 update and got rebooted. So that may have caused things to change.

[image]

I recall that I just had updated the firmware on the device before via VictronConnect, so perhaps the cause was that the gui-v2 was still working with outdated info.

I don't mind if you close this issue and we recreate a new one if it happens again.

chriadam, 2024-11-04:
Ok, thanks, closing for now.  Note that #1301 might be related, but it's a slightly different case I think.

closed 2024-11-04 by chriadam