victronenergy/gui-v2 #1627: Wasm: Onboarding always appears for VRM read-only systems

blammit, 2024-10-24:
Currently, the onboarding screen will always show on start-up when run via wasm on a VRM read-only system. This is because the system is unable to write to `/Settings/Gui2/OnBoarding` to clear the onboarding flag.

Need a straightforward way to determine whether the system is read-only.

One possible solution is to set a cookie in this case, instead of writing a value via MQTT.

mpvader, 2024-10-25:
And another one is to get a parameter from VRM telling the wasm that its read only

blammit, 2025-05-01:
Solution will be to get a parameter from VRM. Setting to blocked.

AlexImb, 2025-05-01:
I'll take care of adding the parameter from VRM next week.

blammit, 2025-05-01:
@AlexImb actually, there is a setting value provided by MQTT already, so looks like this is not needed.

(See https://github.com/victronenergy/gui-v2/issues/560)

chriadam, 2025-05-23:
Merged to main, closing as fixed.

closed 2025-05-23 by chriadam