victronenergy/gui-v2 #1637: VE.Bus AC Input 1 and 2 issues

mpvader, 2024-10-28:
@vejpasop I encountered this by accident, while looking into an unrelated issue where the current limit of EVCS can't be set via gui-v2. Please take this over from me, test more completely and propose how it should be work + list of test conditions.

(Possible) issues that I've seen: 

### 1. Both the input current limit entries shown in the list have the same name.

Whats expected instead is that they have different names; or we show just one.

Twice the same name in the control card:
[image]

Twice the same name shown in details page:
[image]

### 2. the 50A is shown as editable but its not

Better to show it as a non-editable version right from the get go and a toast explaining why when user clicks that.

[image]

### 3. Notification text needs amending. 

It now says "This setting is disabled. Possible reasons are "Overruled by remote" is not enabled or an assistant is preventing the adjustment. Please, check the inverter configuration with VEConfigure". In 99% of cases, its disabled by design by the installer, lets not tell end-users to go into VEConfigure. Suggested new text: "This current limit is fixed in the system configuration, it is not possible to adjust it".

blammit, 2024-10-29:
PR https://github.com/victronenergy/gui-v2/pull/1643 fixes this.

Note that this only works if the vebus service is the primary one on the system, according to `com.victronenergy.system/VebusService`. Otherwise, it is assumed the `/SystemSetup/AcInput<X>` cannot be applied for this vebus service, and it will still just show "Input current limit" instead of something more specific.

vejpasop, 2024-10-29:
Successfully tested on Venus V3.50~42.  
[image]
[image]
[image]
[image]
[image]
[image]

izak, 2024-10-31:
There is still an issue here. After this change, for acsystem services (Multi-RS) the mode is no longer adjustable.

[image]

PR #1675 (2024-11-01): Be less restrictive when disallowing mode and current limit changes
If /ModeIsAdjustable or /CurrentLimitIsAdjustable is not available, then allow the mode or current limit to be adjusted.

Fixes #1637
commit 6c39882e (2024-11-01): Be less restrictive when disallowing mode and current limit changes

If /ModeIsAdjustable or /CurrentLimitIsAdjustable is not available,
then allow the mode or current limit to be adjusted.

Fixes #1637
closed 2024-11-01 by blammit