victronenergy/gui-v2 #1655: PV inverters: Add IP address manually changes not saved

mr-manuel, 2024-10-29:
When manually adding an IP address under Settings -> PV inverters -> Add IP address manually -> Click on + icon, then the IP default address is added, but the changes to the IP entries are not saved.

[image]

mpvader, 2024-10-29:
This means that its not possible to manually add pv inverters?

so effects all systems with pv inverter outside subnet/vlan.

no problem for pv inverters in same sub/net vlan - since they’ll be found by normal scan.

right @mr-manuel ?

closed 2024-10-30 by blammit

mr-manuel, 2024-10-30:
Yes that is correct. Only relevant for PV inverters in another subnet/vlan.

PR #1659 (2024-10-30): PV inverter settings: save changes to manually-added IP addresses
Need to save changes to IP addresses text fields.

Fixes #1655
commit 5aebd048 (2024-10-30): PV inverter settings: save changes to manually-added IP addresses

Need to save changes to IP addresses text fields.

Fixes #1655