victronenergy/gui-v2 #1661: Ukrainian translation of battery and settings menu header

dirkjanfaber, 2024-10-30:
In Ukrainian the word "Battery" no longer gets translated into  "Батарея". See https://github.com/victronenergy/gui-v2/blob/main/i18n/venus-gui-v2_uk.ts#L4425

Strangely enough this is not problem for other languages, which correctly translate the term. 
[image]

And before v3.50~37 it was working correctly:
[image]

Feel free to test on https://vrm.victronenergy.com/installation/102195/dashboard

And also noticing that, when you switch translations, the menu header doesn't get updated until you re-enter the settings menu.
[image]

podarok, 2024-10-30:
Tnx

imval, 2024-10-31:
There is a similar issue for the French translation with the word "Automatic" that is not correctly translated in the UI in a few places, even though the key is present in Poeditor (the translation platform).
[image]
[image]

Dirk-Jan asked me to add this in this issue.

podarok, 2024-10-31:
title seems like not only Ukrainian related @dirkjanfaber 
I have no rights to change it, on you

chriadam, 2024-11-01:
The id for the "Battery" string is "common_words_battery", but this is missing from the locale-specific .ts files in the repository, although it IS present in the locale-specific .ts file in the build artifacts directory, from which the .qm files are generated... Interestingly, the translation is marked as "unfinished" there!

```
    <message id="common_words_battery">
        <location filename="../../gui-v2/components/CommonWords.qml" line="83"/>
        <source>Battery</source>
        <translation type="unfinished">Батарея</translation>
    </message>
```

chriadam, 2024-11-01:
I think we're running lrelease "too soon" i.e. on a version of the .ts which hasn't been properly updated by lupdate.
We have some dependency logic in the CMakeLists.txt file to try to enforce proper ordering, but it seems like this might be failing.

dirkjanfaber, 2024-11-01:
The cause for the `unfinished` marking seems to come from the fact that some translations are marked as "fuzzy" within poeditor.   Even though that seems to be unrelated, I will ask the translators to check these fuzzy ones too and unmark the fuzzy checkbox if they have proofread it.
The rabbit hole on fuzzy translations can be found [here](https://poeditor.com/kb/fuzzy-translations). 🐰

dirkjanfaber, 2024-11-01:
Did test with the build from https://github.com/victronenergy/gui-v2/actions/runs/11626874251. Though "Battery" is correctly translated for Ukrainian, it isn't the case for "Automatic" in French. 

[image]

While it is translated correctly on the ethernet configuration. 😕 
[image]

Can it be that only one instance of the term actually gets translated?

chriadam, 2024-11-04:
Thanks, closing as fixed.

closed 2024-11-04 by chriadam