victronenergy/gui-v2 #1662: Solar detail pages show errors for spurious MPPT RS

mpvader, 2024-10-30:
you can use this test system: 

https://vrm.victronenergy.com/installation/55013/overview

There are no errors, so the count is wrong:

[image]

mpvader, 2024-10-31:
This one should be simple to fix @blammit ?

and many people notice this; pls prioritise this one

chriadam, 2024-10-31:
On our system, the path: com.victronenergy.solarcharger.ttyUSB1/Alarms/LowVoltage etc don't exist, and so the "isValid" check (correctly) fails, and no error is shown.  But, on other systems, the path might exist, but the value might be zero (i.e. "this alarm isn't set").  So we need to update the condition to not just check isValid but also whether value != 0, I guess.

chriadam, 2024-11-01:
Fixed.

closed 2024-11-01 by chriadam