victronenergy/gui-v2 #1665: Fallback or translations of startpage -> display don’t work?

mpvader, 2024-10-31:
this is german,

english does work for me. And german too.. so can’t reproduce. And I expect that soon this will be translated - if you can’t reproduce / don’t see something special, then pls close it as non-reproducible. If there are further reports we can always re open it.

[image]

mpvader, 2024-10-31:
someone else having it all the time:

[image]

mpvader, 2024-10-31:
updated: 
[image]

chriadam, 2024-11-01:
I believe this is (effectively) a duplicate of #1661

chriadam, 2024-11-04:
@dirkjanfaber do you know, was this one also verified to be fixed in v3.51~2?  If so, will close.

dirkjanfaber, 2024-11-04:
Just tested with Romanian and this page. Looks good! Closing. 

[image]

dirkjanfaber, 2024-11-06:
I think we need to re-open this, @chriadam 

[image]

Of course it needs to be translated properly, but as long as the translation isn't there yet, it should show the English variant and not the key. 

https://github.com/victronenergy/gui-v2/blob/main/i18n/venus-gui-v2_sv.ts#L8320

philipptrenz, 2024-11-06:
Another find for the Spanish language

See translation here: https://github.com/victronenergy/gui-v2/blob/main/i18n/venus-gui-v2_es.ts#L8437

[image]

chriadam, 2024-11-08:
I should be able to simply install the english catalogue as a fallback catalogue, in case some translation is missing from a non-english translation catalogue in some version.

mpvader, 2024-11-12:
Is there more remaining work for this one @chriadam ?

chriadam, 2024-11-13:
No, closing as fixed.

closed 2024-11-13 by chriadam