victronenergy/gui-v2 #1667: Device List: "Remove disconnected devices" button shows for usb devices reconnected on different port

blammit, 2024-10-31:
When viewing the Device List, if you unplug a usb device and reconnect it to a different usb port, then the device will appear to be re-connected in the list, but the "Remove disconnected devices"  appears, and clicking it has no effect. 

This happens because when the device is connected to a different usb port, it gets a different service uid compared to before. Perhaps gui-v2 can retain some other device ID to check if it is the same device, instead of just relying on the service uid to do this.

ReinvdZee, 2026-09-10:
@blammit @chriadam 
Is this still reproducible? If not, please close it.

blammit, 2026-09-11:
We haven't changed the implementation so I'm sure this still applies.

(The issue is that the service UID is used to identify all services in this list, but in this case, the UID has changed even though it belongs to the same device. To fix this we need some special handling to recognise that the same device has been reconnected with a different UID, and avoid using just the UID to test whether an entry is already in the list.)

ReinvdZee, 2026-09-24:
@blammit 

We discussed in the meeting today that it would be difficult (at least in some cases) for the UI to determine whether the device plugged in is the same as the device that was just removed. 

From the slack discussion and the screenshot above, it looks like the UI already handles this. The device disappears and comes back in the same entry in the device list. So in this case the "Remove disconnected devices" actually does nothing. Can that be resolved? I.e. can we make sure the "Remove disconnected devices" button goes away when there are no disconnected devices in the list? It feels like a small bug when the button is there but doesn't do anything.