victronenergy/gui-v2 #1679: Missing Grid in GUI 2

tmonxx, 2024-11-01:
Hi.

After switching to GUI 2 our Grid is shown as diconnected…
GUI 1 is ok.
Some ideas?
[image]
[image]

blammit, 2024-11-03:
Sounds like the same issue as https://github.com/victronenergy/gui-v2/issues/1626.

If you could add victron.vrm@qinetic.com.au to the users for your VRM installation, I can verify whether my patch fixes the issue for you.

tmonxx, 2024-11-04:
> Klingt nach dem gleichen Problem wie #1626 .
> 
> Wenn Sie victron.vrm (at) qinetic.com.au zu den Benutzern für Ihre VRM-Installation hinzufügen könnten, kann ich überprüfen, ob mein Patch das Problem für Sie behebt.

Thank You. You are in…

blammit, 2024-11-07:
Danke schön, I tested the https://github.com/victronenergy/gui-v2/issues/1626 fixes on this installation, and the issue is fixed now. The changes were released with VenusOS v3.52~2.

closed 2024-11-07 by blammit

tmonxx, 2024-11-08:
Thank You for Your help.