victronenergy/gui-v2 #1684: ESS Scheduled Charge Level says Inactive when it is Active

mpvader, 2024-11-03:
https://community.victronenergy.com/t/3-51-2-gui-v2-ess-scheduled-charge-level-says-inactive-when-it-is-active/10079

chriadam, 2024-11-04:
Created a fix for a bug I found, related to this.  However, as I commented in the PR:

Unfortunately, this doesn't actually fix the issue. I tested with installation 402872.
I double checked gui-v1 code, and gui-v2 matches that. But, it seems like the com.victronenergy.system/Control/ScheduledSoc value remains invalid / value undefined, even if there is a valid schedule according to e.g. com.victronenergy.system/Settings/CGwacs/BatteryLife/Schedule/Charge/1 etc.

Maybe @ReinvdZee or @izak could comment on this - when should the /Control/ScheduledSoc value be valid?

chriadam, 2024-11-04:
I merged a fix for a bug in gui-v2 code, but even with that code the issue still exists (both in gui-v2 and in gui-v1).  It looks to me like a venus-platform issue (i.e. perhaps it isn't correctly updating the value of com.victronenergy.system/Control/ScheduledSoc when it should)?

izak, 2024-11-04:
This is in systemcalc (not venus platform). Reasons why Scheduled charging will not activate:

1. The battery SOC is not known (eg no battery monitor selected, selected battery monitor not present, etc)
2. The ESS mode is set to Keep Batteries Charged
3. DynamicEss is currently active (this has higher priority)
4. No matching schedule (make sure time zones are not interfering with things)

chriadam, 2024-11-04:
Ah!  Interesting.  I wonder if it's worth adding that information to the drilldown from the "Scheduled charge levels" setting, to help the user understand why it might show as "inactive" even if some valid schedules are specified.

On the other hand: I suspect that the bug actually IS fixed now, and I just looked at an installation for which the schedule was valid but not active, currently.  So, I think I can close this bug as "fixed".

closed 2024-11-07 by blammit