victronenergy/gui-v2 #1710: Improve error messaging / information in VRM use case

mpvader, 2024-11-08:
The system in this screenshot is offline, yet gui-v2 now says connected.

i think I know why, (even though wasm cannot connect to the device itself., it can connect to mqtt cloud) so need to come up with something for this.

[image]

chriadam, 2025-03-03:
I have created the above PR which exposes the heartbeat state as additional error information on the splash view.  It also returns the user to the splash view if the device heartbeat becomes inactive, and then reloads the UI once heartbeat returns.  Let me know if there are some changes wanted to this (e.g. perhaps returning to the splash view is a bit abrupt?)

mpvader, 2025-03-03:
Hi @chriadam can you share a screenrecording showing the result?

chriadam, 2025-03-04:
Here's what it looks like when connecting to a VRM installation where the remote device isn't available (i.e. isn't currently "realtime" connected to VRM, and isn't sending heartbeats to VRM):

https://github.com/user-attachments/assets/6c0ac0ef-b465-4b37-b674-fca701ccea20

And here is what it looks like when an existing wasm gui-v2 instance which is connected to VRM loses its heartbeat and then regains it after some time (i.e. because I rebooted the cerbo in the background):

https://github.com/user-attachments/assets/68ede07a-dd8f-42ff-b691-feb8056de791

mpvader, 2025-03-06:
good improvement, lets put this in with one change: change [4] Connected to ... to :

`[4] Connection to the device has been lost, awaiting reconnection`

chriadam, 2025-03-13:
Merged change, closing as fixed.

closed 2025-03-13 by chriadam