victronenergy/gui-v2 #1714: Settings: clicking on the top row causes the Generator Start/Stop menu option to be triggered

blammit, 2024-11-11:
See thread and video here: https://victrondevelopment.slack.com/archives/C04RYU0UMDE/p1731142689206319

MikeTrahearn-Qinetic, 2024-11-14:
I need access to the victron slack to be able to view the issue details.

blammit, 2024-12-02:
(Access has been provided now)

MikeTrahearn-Qinetic, 2024-12-12:
I have been able to reproduce this.

MikeTrahearn-Qinetic, 2024-12-12:
On investigation it has been found as suspected that the clickable delegates in the Settings list view are not being clipped by the view and are thus clickable through the status bar - AND the navigation bar.
The image below shows this happening clearly as I have made the status and navigation bar slightly transparent, showing the obvious problem in the annotated red areas.
[image]

MikeTrahearn-Qinetic, 2024-12-12:
Adding clipping to the GradientListView in pages/SettingsPage.qml fixes this issue.

MikeTrahearn-Qinetic, 2024-12-12:
- [ ] Test on Desktop
- [ ] Test on WASM
- [ ] Test on Cerbo

PR #1799 (2024-12-16): Fix SettingsPage's GradientListView to add clipping
Clip the GradientListView to prevent delegates that extend beyond the ListView's top and bottom edges from being clickable outside of those boundaries.

This was reproduced and the root cause discovered.

Tested the fix on Desktop, WASM and Cerbo as "can no longer reproduce".

Fixes #1714
commit 5a676a93 (2024-12-16): Fix SettingsPage's GradientListView to add clipping

Clip the GradientListView to prevent delegates that extend beyond
the ListView's top and bottom edges from being clickable outside
of those boundaries.

Fixes #1714
closed 2024-12-16 by MikeTrahearn-Qinetic