victronenergy/gui-v2 #1732: Provide VisibleItemModel that filters out non-visible items

blammit, 2024-11-13:
Provide something like VisibleItemModel from gui-v1 to filter out non-allowed/visible items from settings models, and use this type instead of ObjectModel when creating settings list views (or anywhere else where it makes sense).

There are multiple benefits to this:

- The ListView won't need to create delegates for hidden items
- Key navigation (#1030) won't navigate to hidden items
- Allow ListQuantityGroup and ListTextGroup to only add separators between visible items, instead of needing to add 'visible' configuration properties for items. See #1338.

blammit, 2024-11-13:
If we use this model in the settings list views, I wonder if it can just filter on the `visible` property instead of needing the separate `allowed` property in ListItem. 

The reason `allowed` was added was that dynamic changes to ListItem `visible` caused resizing/layout issues in the ListView. If the items are filtered out from the model altogether, then this shouldn't be an issue.

blammit, 2025-01-15:
> If we use this model in the settings list views, I wonder if it can just filter on the visible property instead of needing the separate allowed property in ListItem.
> The reason allowed was added was [...]

We cannot filter on the `visible` value, because items become non-visible whenever another page is shown on top. So in the settings pages, if a new page is pushed, then the items in the page become non-visible, and then they would be removed from the model and re-inserted when the top page is popped. This would cause the list view position to be reset, and the list view would have to re-position its delegates, etc. So we definitely need the intermediate `allowed` property.

DanielMcInnes, 2025-01-15:
@blammit are these changes still in draft, or ready for review?

blammit, 2025-01-15:
PR: https://github.com/victronenergy/gui-v2/pull/1845

blammit, 2025-02-14:
Merged PR https://github.com/victronenergy/gui-v2/pull/1845

closed 2025-02-14 by blammit