victronenergy/gui-v2 #1735: Hide position setting for energy meters that don't support it.

ldoesvictron, 2024-11-13:
The VM-3P75CT is configured over VictronConnect and has an internal setting for the meter position. Therefore it doesn't support the position setting in the GUI.

At the moment a position setting does appear in gui-v2, but it doesn't actually do anything. 
[image]

In gui-v1 the position setting is conditional depending on the availability of allowed roles on dbus, see [here](https://github.com/victronenergy/gui/blob/776ec5bc2323baddaa7467ea8d59123832bc5e3a/qml/PageAcInModel.qml#L152).

blammit, 2024-11-20:
Note: this is referring to the PV inverter settings page, which is shown for Energy Meters that are configured to show up as `pvinverter` services.

chriadam, 2025-03-14:
Is there a specific installation I can test this with?  I tried testing with https://vrm.victronenergy.com/installation/394636/dashboard however it seems to have an SMA Solar Inverter and the energy meter must not be configured to show up as a pvinverter, I believe.

Also, interestingly, looking at the raw values for the SMA Solar Inverter device (i.e. settings -> debug & develop -> values -> pvinverter -> 20) we see that the /AllowedRoles path has an invalid value (`--`), even though I believe that this is NOT an energy meter but instead is a PV inverter (or am I wrong about this?).  Should we hide the position setting in this case also, or only in the case where the pvinverter is actually an energy meter (and if so, how can this condition be detected programmatically?)  Note that there appears to be a valid value for the /Position path (`0`) despite the /AllowedRoles value being null/invalid.

chriadam, 2025-03-14:
@ReinvdZee perhaps this is something you might know the answer to ^

chriadam, 2025-05-01:
Hi Martin, please see the discussion in https://github.com/victronenergy/gui-v2/pull/1980 - I'm not sure what to implement in gui-v2 precisely.

mpvader, 2025-05-01:
Unblocked - thanks Martin

chriadam, 2025-05-02:
Updated PR now.

chriadam, 2025-05-07:
Merged to main, closing as fixed.