victronenergy/gui-v2 #1752: Show missing AC Chargers on Gui 2.0 charging (Specifically skylla chargers)

tingles2233, 2024-11-20:
Show AC Chargers such as Skylla Chargers (We have the Skylla-Ip65) show up on the overview page for a quick glance as it does with Gui Mods.

Skylla chargers are used extensively in the boating world (quite often in parallel) to quick charge large battery banks on boats. 

Here is how it showed up in Gui-Mods in V1; 
[image]

It would be great to have the "Active" charging sources show up on the left top side rotating if needed through the two available blocks. 
[image]

In our setup(and most boats) it would show the last two "AC" charging sources in the top left, rotating as they become active; 
1) Shore (AC-IN 2 on Quattro)
2) Generator (AC-IN 1 on Quattro)
3) AC Charger (Skylla-IP65)

We also do have 3 MPPT Controllers and our Alternator consuming the "DC" Charging two slots in the bottom left.

tingles2233, 2024-11-20:
Further to update for v1 flow the old gui showed the first block represented as “Generator”(in red) in the screenshot, it would switch between the active input on the Quattro being “Shore” or “Generator”.

the next block down was always the AC Charger.

blammit, 2024-11-25:
@jepefe do you know more about this requirement?

tingles2233, 2024-12-02:
Hi Blammit,

Curious to get a bit more information on what part of the design is not ready for this, and potentially when it is expected to be unblocked. 

Asking as this change affects a lot of boats running Victron setups who would like to upgrade to start using the new interface. 

Thanks in advance!

blammit, 2024-12-03:
@tingles2233 I can't give a timeline for this, but basically the current UI provides access to the AC charger if you click on the Inverter/Charger, and there are design discussions required to decide where to put additional charger information in the Overview page. Perhaps @mpvader can comment more on this.

tingles2233, 2024-12-03:
Thanks @blammit appreciate the update. Understood. I`ll keep this in mind for my setup along with inform others that I know have this same request. 

Please do keep this thread updated with any progress. When/If you require testing let me know.  

Have a great day!

chriadam, 2025-11-06:
Pushing back to v1.3.0, but raising priority as I think we should discuss this early in the v1.3.0 cycle and make a decision (either to close the task as WontFix or to choose which additional info to show on the Overview widget).

tingles2233, 2025-11-06:
Thanks Chriadam, I hope it is added as. I know there is lots of talk here in the boating world about seeing it natively on the cerbo as we could before without having to dig into the device list to see it charging.