victronenergy/gui-v2 #1754: DVCC temperature selection menu: Use customname instead of product name

ReinvdZee, 2024-11-22:
In the DVCC menu, there is an option to select which temperature sensor is used for STS:

[image]

However, for temperature sensors connected to the SmartShunt, this menu shows the product names of the SmartShunts which can be selected:
[image]
For systems with multiple SmartShunts with temperature sensors, it is not possible for the user to distinguish between SmartShunts in order to select the desired SmartShunt to use the temperature data from.

The Solution is to show the customname of the devices instead of the product name.

This issue can be reproduced on [this sytem](https://vrm.victronenergy.com/installation/323213/dashboard), navigate to Settings->DVCC->Temperature sensor (STS needs to be enabled).

chriadam, 2025-02-25:
@ReinvdZee could you help to test PR#1917 please?  I checked with the installation you mentioned, and see "SmartShunt" in the radio button list now (which I assume is a custom name, since it's not the entire product name).

chriadam, 2025-03-03:
PR merged.  Closing as fixed - can be reopened if there's a case which doesn't work still :-)