victronenergy/gui-v2 #1761: Reduce CPU load of Brief side panel

blammit, 2024-11-26:
The Brief side panel has a high CPU load due to the graph rendering. For some users this means the "System load high, closing the side panel to reduce CPU load" warning is regularly displayed. 

Ideally the message would be displayed less often, and ideally the CPU load of the side panel could be reduced. This needs some investigation into the graph rendering.

Example community reports:

https://community.victronenergy.com/t/3-50-unexpected-message-about-closing-the-side-panel/12419/3
https://community.victronenergy.com/t/new-gui-warning/10023/3

chriadam, 2025-03-25:
I have now merged a change which should reduce CPU usage by a handful of percentage points, according to my testing.  However, the largest consumer of CPU usage in that view is actually the MultiCircularGauge.  When setting all levels to "none" the CPU usage on the office cerbo GX reduces from about 50% down to 20%.

I will look into whether any improvements in the rendering of those CircularGauge PathArcs is possible, to reduce rendering cost...

chriadam, 2025-03-25:
I was wrong.  It's not the 4 gauges.  It's actually just the different demos I'm using to test this.  With the "Motorhome/Boat 2" demo, the side panel has 5 different graphs (solar yield, grid, alternator, ac loads, dc loads) and the combined cost sits at around 53% on average.  I'm not sure what we can do about this.  It's possible that rewriting the graphs as shaders would be faster, but it's most likely that it will be the same cost (as seen with the circular gauge rewrite effort) as it's more to do with the number of pixels being rendered each frame than the specific mechanism being used to do the rendering (Shape vs hand-crafted shader).

Otherwise, a simple solution might be to offer settings to allow the user to disabled specific graphs on the panel in order to reduce CPU usage, I suppose?

chriadam, 2025-03-25:
Unassigning myself for now - it's lower priority than other tasks, and some improvement was made already.