victronenergy/gui-v2 #1763: AC input current gauge max limit is NaN.

ReinvdZee, 2024-11-26:


blammit, 2024-11-27:
Looking into this, the NaN value is being shown for this path: `com.victronenergy.settings/Settings/Gui/Gauges/Ac/In/1/Current/Max`. This value is not present on the installation, but I'm not sure why?

> AC input 2 is connected: com.victronenergy.settings/Settings/Gui/Gauges/Ac/AcIn2/Consumption/Current/Max.

This path is used for the "Maximum current: AC in 2 connected" value further down the page.

ReinvdZee, 2024-11-27:
@blammit You're right.

The setting is missing on some systems because there was a setting name duplicate in systemcalc. I fixed it: https://github.com/victronenergy/dbus-systemcalc-py/commit/0fb1a99f0ed932a731cdc56d919830b787bfa8b7

And verified on the ESS system that all setting paths show up correctly.