victronenergy/gui-v2 #1773: GUI V2 displays the wrong inverter

seamaster101, 2024-12-02:
I have the following setup:
Quattro II 
Phoenix 
Ekrano that just replaced my RPI Venus previos setup
Two alternators etc. 
The issue is that I have the overview screen displaying the 1200VA Phoenix inverter that I use to power my refrigerator, rather than my Quattro II main inverter. For that reason I don’t see AC inputs or my main inverter AC loads on the overview screen which I use all the times. 
I searched through the settings but did not find way of selecting which inverter to display, or even better display them both. I also searched forums etc. with no luck.  I even went to the extend or renaming both inverters in case the systems displays the first one in chronological order, but apparently that was not the case…
Also please note that the overview page is not displaying the custom names but rather the generic names which is confusing when more than one device of a type exists in the setup e.i. two alternators. 
Is there way to prioritize that I’m missing or this is something that is not been considered in the design?
[image]
[image]

blammit, 2024-12-03:
@seamaster101 If there are multiple `inverter` services then the UI will pick the one with the lowest device instance. If you look at the "Device" page for each inverter, then you can see its device instance number.

@ReinvdZee Is there a way to configure the device instance for inverter/chargers? 

> Also please note that the overview page is not displaying the custom names but rather the generic names which is confusing when more than one device of a type exists in the setup e.i. two alternators.

If there are multiple alternators then it shows a summary for the alternators, i.e. the combined power.

seamaster101, 2024-12-03:
@blammit i just check wha you suggest but it is not exactly correct. My Quattro II has VRM instance of  289 and my phoenix 1200 has VRM instance of 291. 
according to what you suggested I should be seeing the Quattro but I’m seeing the Phoenix. 
With regard to the alternators this does not makes sense what so ever. The alternators are separate devices and could be used for different tasks thus combing them does not accomplish much in these cases. So example - one alternator charges house batteries the other charges starting batteries. Why would anyone want them combined? Even if they are charging the same battery, users would like to know which alternator is doing what part of the job. Food for thought….
[image]
[image]
[image]

seamaster101, 2024-12-03:
@blammit @ReinvdZee
There is another test that I just did:
I disconnected the Phoenix inverter and left just the Quattro II connected. Then I restarted the Ekrano and my expectation was that I will get my main inverter show properly on the overview screen. Unfortunately something quite unexpected and not correct is happening:
it shows the inverter is Off on the overview screen but shows the inverter is in float mode on the device list screen. 
there is something not right. I think this needs to be investigated as it is obvious misbehavior. Maybe there is underlying bug???
[image]
[image]

blammit, 2024-12-04:
@seamaster101 Actually, what I said earlier about picking the lowest instance is something that applies to other areas where we need to choose between multiple inverter/chargers, but that's not the issue in this case. The Inverter/Charger box should show the overall system state. If you could add access for victron.vrm (at) qinetic.com.au to your installation, then I can check what is going on there.

blammit, 2024-12-05:
@seamaster101 Talking with others about this, it appears that you have plugged the Multi in with a USB cable and not directly into the Ekrano. Can you check?

Technical detail: the state shown in the Inverter/Charger box is an overall system state that comes from the system `/SystemState/State`, which is the state of the built-in port. So if you plug the inverter in directly into the Ekrano, instead of via USB, then the Inverter/Charger box should show the expected state.

seamaster101, 2024-12-06:
Yes it is plugged in into USB. via Victron adapter as before I had RPI install. Should it work over USB as well? i have no enough cable length to plug in direct and rerouting new cable is major undertaking...

blammit, 2024-12-06:
Currently the system state is taken from the state of the device plugged into the built-in port. Perhaps in future there could be an option to choose the USB device or others instead, but it's not supported at the moment.

closed 2024-12-06 by blammit