victronenergy/gui-v2 #1778: Can't change VRM device instances locally

ReinvdZee, 2024-12-03:
On WASM, the VRM Device instances of the connected devices can be changed properly in this menu:

[image]

On the local gui, the connected devices are not present in this menu:

[image]

This can be tested on [this](https://vrm.victronenergy.com/installation/473156/dashboard) system, running v3.60~7

blammit, 2024-12-09:
It looks like the qtvirtualkeyboardplugin  is not installed correctly in VenusOS v3.60~7. I can see that `/usr/lib/plugins/platforminputcontexts` is missing, but that was present in VenusOS v3.50. @jhofstee would you know what is happening here?

blammit, 2025-02-14:
https://github.com/victronenergy/gui-v2/pull/1905 fixes the names in the list, and also ensures the `/CustomName` for the entry is shown if available.

commit d1985bec (2025-02-14): Fix loading of device names in VRM device instances

Instead of using Global.deviceModelsForClass() to find device objects,
fetch device instances from AllDevicesModel, to guarantee that a
device object is found if available on the system.

This was not previously possible as the "device class" from the
/ClassAndVrmInstance value did not necessarily translate to a service
type; for example, a tank or temperature sensor used "analog" as the
device class. However, the underlying system now provides the service
type as the device class, which means that AllDevicesModel can now be
used to find the [... cut]
commit afa2672a (2025-02-14): Fix loading of device names in VRM device instances

Instead of using Global.deviceModelsForClass() to find device objects,
fetch device instances from AllDevicesModel, to guarantee that a
device object is found if available on the system.

This was not previously possible as the "device class" from the
/ClassAndVrmInstance value did not necessarily translate to a service
type; for example, a tank or temperature sensor used "analog" as the
device class. However, the underlying system now provides the service
type as the device class, which means that AllDevicesModel can now be
used to find the [... cut]
commit cf036f81 (2025-02-14): Fix loading of device names in VRM device instances

Instead of using Global.deviceModelsForClass() to find device objects,
fetch device instances from AllDevicesModel, to guarantee that a
device object is found if available on the system.

This was not previously possible as the "device class" from the
/ClassAndVrmInstance value did not necessarily translate to a service
type; for example, a tank or temperature sensor used "analog" as the
device class. However, the underlying system now provides the service
type as the device class, which means that AllDevicesModel can now be
used to find the [... cut]
commit 9b96552b (2025-02-14): Fix loading of device names in VRM device instances

Instead of using Global.deviceModelsForClass() to find device objects,
fetch device instances from AllDevicesModel, to guarantee that a
device object is found if available on the system.

This was not previously possible as the "device class" from the
/ClassAndVrmInstance value did not necessarily translate to a service
type; for example, a tank or temperature sensor used "analog" as the
device class. However, the underlying system now provides the service
type as the device class, which means that AllDevicesModel can now be
used to find the [... cut]
closed 2025-02-17 by blammit