victronenergy/gui-v2 #1780: Local OSK not showing up

ReinvdZee, 2024-12-04:
On Venus OS 3.60~7, the OSK is not showing up on the local GUI. Only a blinking cursor is showing. 
To reproduce, try setting a custom name for a device or the GX's custom name (in settings -> system setup)

[image]

This is working properly in Venus OS 3.52.

blammit, 2024-12-09:
It looks like the qtvirtualkeyboardplugin  is not installed correctly in VenusOS v3.60~7. I can see that `/usr/lib/plugins/platforminputcontexts` is missing, but that was present in VenusOS v3.50. @jhofstee would you know what is happening here?

closed 2024-12-09 by jhofstee

jhofstee, 2024-12-09:
Thanks, I split the plugins into separate packages, so unused plugins can be left out to free up diskspace. The stub shared library for the virtual keyboard was also left out while it is needed, so I added it back.