victronenergy/gui-v2 #1822: Add caption property to ListItem

blammit, 2025-01-08:
There are various list items with caption text, which are implemented by adding custom Labels to the `bottomContent`, like this:

```
bottomContentChildren: [
    PrimaryListLabel {
        width: Math.min(implicitWidth, listItem.maximumContentWidth)
        topPadding: 0
        bottomPadding: 0
        color: Theme.color_font_secondary
        text: "Custom text label"
    }
]
```

Remove these and add a configurable caption to the base ListItem instead.

closed 2025-02-12 by DanielMcInnes

MikeTrahearn-Qinetic, 2025-02-12:
Having a chat with myself here: "Could the property string caption not be a property alias to the as-yet un-id-entified Label? The only reason I would say to keep it as as string, is that the entire bottomContent is aliased and could be wiped out, making a property alias to the caption Label a bit "dangling" if not invalid."

DanielMcInnes, 2025-02-12:
> Having a chat with myself here: "Could the property string caption not be a property alias to the as-yet un-id-entified Label? The only reason I would say to keep it as as string, is that the entire [...]

Could be. I had it as an alias originally, Bea asked me to change it to a property

MikeTrahearn-Qinetic, 2025-02-12:
Probably best it is left as a property for at least the reason that the bottomContent could be replaced.

commit b8315591 (2025-01-27): Add caption property to ListItem

Fixes #1822
commit 661eb09d (2025-01-28): Add caption property to ListItem

Fixes #1822
commit 363e317e (2025-02-11): Add caption property to ListItem

Fixes #1822
commit 0b2a4736 (2025-02-11): Add caption property to ListItem

Fixes #1822